freetronics/DMD2 #5: Attempted ESP8266 Port

h4rm0n1c, 2015-07-08:
I'm not sure if you're interested or not, but since you're the author, I thought I'd at least bring it to your attention.

I used this library a few years ago to create a NTP synced clock with a DMD, and it turned out pretty well, worked just as I wanted it to, the library was a huge help, fast forward a few years and this new ESP8266 based branch of the Arduino IDE has been released, (or it can be installed as a board manager package).

https://github.com/esp8266/Arduino

I thought it would be really cool to have the ability to make really cheap wifi dot matrix displays with the ESP8266 like this, and with a price around a few dollars per module, it's really attractive.

Unfortunately my attempt to port your wonderful library has been amateurish and ultimately not successful, I just hope I haven't made too much of a mess of things in the process :(

There are a few libraries included with the ESP8266 package that seem to use SPI on this platform successfully, along with a number of GPIO pins, I tried looking at those for some ideas, but I fear that may have been more of a mistake than anything.

The work I've done is on my own fork, I know it's a lot to ask, but do you have any advice?, I'm not sure what to do any more.

I apologize in advance if using the issues tracker is not your preferred point of contact or if you find this to be an irrelevant topic to your tracker.

h4rm0n1c, 2015-07-09:
I've actually managed to get this working.

Needless to say, the SPI library in the ESP8266 branch of the Arduino IDE needs some work.

https://www.youtube.com/watch?v=B7r0CkXp2Bk

I'll try and clean this up, maybe you'll have a pull request to add full ESP8266 support soon!

closed 2015-07-09 by h4rm0n1c

projectgus, 2015-07-11:
Looks great! Looking forward to the pull request, thanks.