freetronics/DMD2 #6: Ghosting

jrossouw, 2015-07-13:
Hi,

I have found that with some panels the new library has issues where some LEDs that should be off are not 100% off (dimly lit) and also some LEDs that should be on have higher brightness than the remaining. I initially thought it might be the SPI clock rate, but slowing it down (or speeding it up) did not change anything. Also going to SoftDMD resulted in exactly the same behavior. Any ideas? Could it be the latch timing on the DMD shift register? I see there is a comment in the code about leaving it is digital write to get a decent latching time. Could it be the latching time is still too slow? Should this code be writing directly to the ports to speed up the latch pulse?

Regards,
Johan

projectgus, 2015-08-02:
Hi Johan,

Sorry for the slow reply. I haven't seen this with the DMDs that I've tested with. Which DMD product are you connecting to? Have you tested with the old DMD library as well?

Is the uneven brightness between horiztonal rows, between vertical columns, or follow any other pattern?

Do you have enough current capability in the power supply feeding the DMD? They do require a lot of current, if the 5V output is drooping or glitching down then it may lead to uneven brightness.

Angus

jrossouw, 2015-08-03:
Hi Angus,

I just received (as in on Friday) a different set of LED panels from a 
different supplier. They don't show the problem and work as expected - 
so I suspect it is a hardware problem. You can close this ticket.

Thanks,
Johan

projectgus, 2015-08-03:
OK, good to hear. Thanks Johan.

closed 2015-08-03 by projectgus