freetronics/DMD2 #9: Font Cyrillic

alexzzzzz, 2015-11-03:
Support for Russian fonts. DMD but supports a number of disadvantages. Is it possible to make the support of Cyrillic in the library DMD2?

alexzzzzz, 2016-01-21:
The panel is connected on your circuit. With a library of DMD Russian letters are displayed well, but with a library DMD2 displayed only Latin characters. I assume that the library DMD2 not correctly calculated the size of the character with 0x80 - 0xFF

h4rm0n1c, 2018-06-21:
This library is beta, as far as I know, there is no support for alternative character sets at this stage.
My own level of expertise as maintainer leaves me high and dry in terms of being able to implement this myself, but if I receive a pull request implementing such functionality I will definitely review, test and merge if possible.

closed 2018-06-21 by h4rm0n1c