freetronics/DMD2 #13: Pin Explanation

trishmapow, 2016-05-02:
Hi, I want to use the DMD with my Raspberry Pi and there are currently no libraries available. I would like to experiment with connecting it to the GPIO. Could you please explain the function of the pins (CLK, R, nOE, SCLK, B, A)?

smching, 2016-05-05:
DMD uses HUB12 interface which is similar to HUB08, you should look for HUB12 protocol.

More information about the pin function is available here:
http://forum.freetronics.com/viewtopic.php?t=5580

You should read this article as well:
http://cl.hatenablog.com/entry/picari-ledmatrix-raspberrypi

Traslate to English:
http://translate.google.com/translate?langpair=auto%7Cen&u=http://cl.hatenablog.com/entry/picari-ledmatrix-raspberrypi

mohitsachan33, 2016-08-06:
hi
can you help me how to connect nodemcu with DMD and how to program it.
thanks.

closed 2018-06-21 by h4rm0n1c