freetronics/DMD2 #15: Brightness and Drawing

sanchez, 2017-01-20:
I am implementing my own dmd library (because I am avoiding the arduino library). And I was wondering two things. Firstly if the brightness works correctly, when I attempted to do brightness I got a ghosting effect. Secondly, if I lower the refresh rate on the display, every fourth row is brighter than the previous. Is there a reason for this?

sanchez, 2017-01-20:
I fixed the fourth row problem by removing continuous scanning and went with incremental scanning similar to what has been implemented here. However I still have problems with the brightness adjustment

h4rm0n1c, 2018-06-21:
These displays are full of quirks due to their simple hardware and somewhat demanding timing requirements. My only suggestion is trial and error, that's how I got the ESP8266 support working.

Since this is not actually an issue with this library and more a request for technical info that is already available elsewhere, I'm closing this issue due to it being technically off topic.

closed 2018-06-21 by h4rm0n1c