freetronics/DMD2 #20: DMD2 Library not working in ESP8266 Station Mode

NeerajShrivastava04, 2017-08-20:
Hi,
First of all I want to thank you for such a wonderful library.
I am working on a proto in which I need to run the P10 Modules with ESP8266 and need to show the data fetched from the web. I am wondering if anyone else has faced this issue so far or not but when putting the ESP8266 in STATION mode along with DMD2 library, then my NodeMCU is getting rebooted again and again. Its getting rebooted at the point where I have setup the WIFI mode in my setup() function. The same works fine in case if the ESP is being configured in ACESS POINT Mode.

I have decoded the exception with the help of exception decoder but I couldn't figure out much in that.
Please find below my code.

`#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <SPI.h>
#include <DMD2.h>
#include <fonts/SystemFont5x7.h>
#include <fonts/Arial14.h>
#include <ESP8266WebServer.h>

/////////////////// Constants ////////////////////
const char *ssid =  "Neerajshri";     // change according to your Network - cannot be longer than 32 characters!
const char *pass =  "neeraj@123"; // change according to your Network

const uint8_t *FONT = Arial14;
//const uint8_t *FONT = SystemFont5x7;

const char *MESSAGE = "abcdefghijklmnopqrstuvwxyz";
//char message[5] = "abcd";

HTTPClient http;
const int WIDTH = 1;

SPIDMD dmd(WIDTH,1);  // DMD controls the entire display
DMD_TextBox box(dmd);  // "box" provides a text box to automatically write to/scroll the display

//int indexStop =0;

// You can change to a smaller font (two lines) by commenting this line,
// and uncommenting the line after it:

void setup() {
  Serial.begin(115200);

  dmd.setBrightness(255);
  dmd.selectFont(FONT);
  dmd.begin();

  Serial.println(F("Booting...."));
  int retries = 0;
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  while ((WiFi.status() != WL_CONNECTED) && (retries < 10)) {
    retries++;
    delay(500);
    Serial.print(".");
  }

  if (WiFi.status() == WL_CONNECTED) {
    Serial.println(F("WiFi connected"));
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
  
  Serial.println(F("Ready!"));
  Serial.println(F("======================================================")); 

}

void loop() {
  // put your main code here, to run repeatedly:

   const char *next = MESSAGE;
  while(*next) {
    Serial.print(*next);
    box.print(*next);
    delay(200);
    next++;
  }
}

`

Also, Please find below the output of exception decoder.

> 0x40101e46: pp_post at ?? line ?
0x40235f80: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4020fa35: wifi_softap_stop at ?? line ?
0x4010020c: _umm_free at C:\Users\nshrivas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x40105a78: ets_timer_arm_new at ?? line ?
0x40209316: pp_enable_idle_timer at ?? line ?
0x40235f80: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40105e06: spi_flash_erase_sector at ?? line ?
0x401075dc: pvPortZalloc at C:\Users\nshrivas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x4021f578: wifi_param_save_protect_with_check at ?? line ?
0x4021f561: wifi_param_save_protect_with_check at ?? line ?
0x4021f649: system_param_save_with_protect at ?? line ?
0x4021f48d: wifi_set_broadcast_if at ?? line ?
0x4021f4da: wifi_set_opmode at ?? line ?
0x40203f5d: ESP8266WiFiGenericClass::mode(WiFiMode) at c:\users\nshrivas\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2439
0x40206d2c: Print::print(char const*) at C:\Users\nshrivas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 64
0x402025f0: setupAP(char*, char*, char*, char*) at c:\users\nshrivas\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40206d44: Print::println() at C:\Users\nshrivas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 64
0x4020282e: setup at c:\users\nshrivas\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x402078fc: loop_wrapper at C:\Users\nshrivas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\nshrivas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 10

I am using 

- NodeMCU 1.0 (ESP-12E Module) 

- Arduino version is 1.8.2

Would highly appreciate any help in this regard as I am running very short of time for completing this POC and I totally stuck over here.

Thanks & regards,
Neeraj

helmisusanto, 2017-12-09:
Still get problem?

Esp8266 still problem if you define dmd first,

You should define your wifi configuration first then you define dmd.

Just change place

And all will working.

helmisusanto, 2017-12-09:
Sorry for incomplete message,

In setup function 

You need to connect wifi First, then you write dmd.begin

Gourishmesta, 2018-02-09:
hi  NeerajShrivastava04 
 which pins of ESP8266 are connected to OE,A,B, CLK,SCLK and for data??

NeerajShrivastava04, 2018-02-09:
Hi I have used the same PINS which are being mentioned in the link below
https://github.com/freetronics/DMD2/pull/7#issue-97282971

Also, the problem which I was facing when I posted this issue is that both DMD2 and ESP 8266 SDK uses  SPI library and because of this a user can't show the data on board in case if ESP is connected to WIFI.
For showing the proper data, user will have to disconnect the WIFI and then will have to initialize the DMD2.

helmisusanto, 2018-02-11:
#define pin_A 16
#define pin_B 12
#define pin_sclk 0
#define pin_clk 14
#define pin_r 13
#define pin_noe 15

#define DISPLAYS_WIDE 1
#define DISPLAYS_HIGH 1
SPIDMD dmd(DISPLAYS_WIDE, DISPLAYS_HIGH, pin_noe, pin_A, pin_B, pin_sclk);  // DMD controls the entire display

P10 sort PIN

NOE | A
-       | B
-       | -
-       | CLK
-       | SCK
-       | R

Correct Is
a | GPIO16
b | GPIO12
- | -
clk | GPIO14
sck | GPIO0
r | GPIO13
noe | GPIO15

Gourishmesta, 2018-02-13:
thank you all for replying 
my issue is solved.

junaid2196, 2018-04-09:
@NeerajShrivastava04 can you help me out with this project....i am unable to display anything on my dmd p10 display...please contact me on my email which is junaidteli922@gmail.com

techsavvy123, 2018-11-24:
> What problem are you facing ??
> Sent from my iPhone
> can u please tell me how did u solve your issue of booting,as i m stuck
> on the same page.

My program is entering into booting mode and i m not able to coming out of
that loop. This is the code which i ve followed.
https://github.com/freetronics/DMD2/issues/20

PerdanaElang, 2021-01-23:
> Hi I have used the same PINS which are being mentioned in the link below
> [#7 (comment)](https://github.com/freetronics/DMD2/pull/7#issue-97282971)
> 
> Also, the problem which I was facing when I posted this issue is that both DMD2 and ESP 8266 SDK uses SPI library and because of this a user can't show the data on board in case if ESP is connected to WIFI.
> For showing the proper data, user will have to disconnect the WIFI and then will have to initialize the DMD2.

How to ? I tried repeatedly to find no results

PerdanaElang, 2021-01-23:
> @NeerajShrivastava04 can you help me out with this project....i am unable to display anything on my dmd p10 display...please contact me on my email which is [junaidteli922@gmail.com](mailto:junaidteli922@gmail.com)

How to ? I tried repeatedly to find no results