freetronics/DMD2 #25: support for esp32

jhonoryza, 2017-10-25:
is there a support for esp32 board?

szymoni, 2017-11-03:
i'm also looking forward to use this library with esp32. Any chances soon?

h4rm0n1c, 2018-06-21:
Not at this stage.

ESP32 was too new when I originally started writing the ESP8266 support, the arduino environment for it was still very early in development and I could not afford the time, money or energy to start on a new platform at the time.

Barring a pull request from someone else, this isn't likely to happen soon. It's definitely something I'd like to see supported myself though as I have plans to use the DMD with an esp32 at some point.

closed 2018-06-21 by h4rm0n1c

learn2007, 2019-08-22:
hi
is there a support for esp32 board now?

sayanlp99, 2020-03-10:
esp32 support arrived yet?