freetronics/DMD2 #26: Problem with ESP8266 Get Request and DMD2 Library

viceplayer, 2018-02-23:
Hello Dear All,

I am using ESP8266(Nodemcu v1.0) to download information from server using simple get Request and DMD2 library to draw them on display.

Problem I have is, that when I call dmd.begin() after some time ESP8266 will become unable to connect to wifi, sometimes it even causes watchdog resets wdt.

I found out that both of them use Interrupts. It might be a problem, I think.

Could you help me to fix it?
[code.zip](https://github.com/freetronics/DMD2/files/1752135/code.zip)

jhonoryza, 2018-04-02:
you need to connect to wifi first then call dmd.begin after that, otherwise it will crashh

h4rm0n1c, 2018-06-21:
It is recommended that you take care of the wireless stack and make sure you're not inadvertently triggering the watchdog timer before starting to use resource heavy peripherals like the DMD.

Much of the esp8266 environment is software implementations of features that otherwise exist in hardware on the AVR platform, this can lead to... quirks.

closed 2018-06-21 by h4rm0n1c