freetronics/DMD2 #29: ESP8266 flip bit

CTWF, 2018-03-24:
Hi.
i need mod dmd2 esp8266 support flipbit to make wiring between row panel more simple

luiz76, 2020-02-28:
Yes! Its so much easier to wiring like this:
[image]

Karo95, 2020-05-29:
flipping panels
open the DMDFrame.cpp file
and search
`void DMDFrame::setPixel(unsigned int x, unsigned int y, DMDGraphicsMode mode)
{
  if(x >= width || y >= height)
     return;`

 and insert for that
`//karo flip begin
  if(height/PANEL_HEIGHT > 1){
	  bool boolpar = ((height/PANEL_HEIGHT) % 2 == 0) ? true:false;
      int delY = y/PANEL_HEIGHT;
      int zvi = y % PANEL_HEIGHT;
        if (((delY) % 2 == 0)&& boolpar || ((delY) % 2 != 0)&& !boolpar ){ 
          zvi = PANEL_HEIGHT-1 - zvi;
          y = (PANEL_HEIGHT*delY)+zvi;
          x = width-1 - x;
        }
    }
  //karo flip end`

h4rm0n1c, 2026-11-11:
Alternate-row panel flipping is now available on master via `setFlipPanels(true)`. #56/#57 were merged through [#72](https://github.com/freetronics/DMD2/pull/72), with follow-up fixes so pixel reads, copying and subframes agree with the flipped layout. Tests include two physical panels on ESP8266 and framebuffer/copy checks; a complete physical multirow wall has not been tested. Closing this feature request as implemented.

closed 2026-11-11 by h4rm0n1c