freetronics/DMD2 #30: esp8266 softdmd now works

jhonoryza, 2018-04-02:
i modified this library with softdmd with esp8266 current arduino firmware, and it works no problem
maybe u can implemented here :)

najeebhall, 2018-04-19:
could you explain what changes you made to get it to work while the esp8266 is connected?

jhonoryza, 2018-04-19:
just comment out the softdmd for esp8266 in DMD.h and DMD.cpp

najeebhall, 2018-04-19:
could you take a look at my issue if possible? https://github.com/freetronics/DMD2/issues/31

h4rm0n1c, 2018-06-21:
The only real reason softdmd is disabled is because the esp8266 has so few resources that anything but using the hardware implementation is probably a bad idea in terms of having available cpu time for other background tasks as well.

I'm happy to change it, but I'd rather have time to test it before I do.

MohamadAlsadi, 2019-07-18:
Can you send me the library after change 
ise34857@gmail.com