freetronics/DMD2 #32: Support for blue pill STM32 (STM32F103C8T6)?

rtek1000, 2018-05-20:
Hello,

Could you add this support?

When attempting to compile for STM32F103C8T6 these errors occur:

> WARNING: library DMD2 claims to run on (avr, sam, esp8266) architecture(s) and may be incompatible with your current board which runs on (STM32F1) architecture(s).
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp: In function 'void TC7_Handler()':
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:99:16: error: 'TC2' was not declared in this scope
> 
>    TC_GetStatus(TC2, 1);
> 
>                 ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:99:22: error: 'TC_GetStatus' was not declared in this scope
> 
>    TC_GetStatus(TC2, 1);
> 
>                       ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp: In member function 'void BaseDMD::begin()':
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:107:19: error: 'TC7_IRQn' was not declared in this scope
> 
>    NVIC_DisableIRQ(TC7_IRQn);
> 
>                    ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:107:27: error: 'NVIC_DisableIRQ' was not declared in this scope
> 
>    NVIC_DisableIRQ(TC7_IRQn);
> 
>                            ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:109:29: error: 'pmc_set_writeprotect' was not declared in this scope
> 
>    pmc_set_writeprotect(false);
> 
>                              ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:110:33: error: 'pmc_enable_periph_clk' was not declared in this scope
> 
>    pmc_enable_periph_clk(TC7_IRQn);
> 
>                                  ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:112:16: error: 'TC2' was not declared in this scope
> 
>    TC_Configure(TC2, 1, TC_CMR_WAVE | TC_CMR_WAVSEL_UP_RC | TC_CMR_TCCLKS_TIMER_CLOCK4); // counter up, /128 divisor
> 
>                 ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:112:24: error: 'TC_CMR_WAVE' was not declared in this scope
> 
>    TC_Configure(TC2, 1, TC_CMR_WAVE | TC_CMR_WAVSEL_UP_RC | TC_CMR_TCCLKS_TIMER_CLOCK4); // counter up, /128 divisor
> 
>                         ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:112:38: error: 'TC_CMR_WAVSEL_UP_RC' was not declared in this scope
> 
>    TC_Configure(TC2, 1, TC_CMR_WAVE | TC_CMR_WAVSEL_UP_RC | TC_CMR_TCCLKS_TIMER_CLOCK4); // counter up, /128 divisor
> 
>                                       ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:112:60: error: 'TC_CMR_TCCLKS_TIMER_CLOCK4' was not declared in this scope
> 
>    TC_Configure(TC2, 1, TC_CMR_WAVE | TC_CMR_WAVSEL_UP_RC | TC_CMR_TCCLKS_TIMER_CLOCK4); // counter up, /128 divisor
> 
>                                                             ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:112:86: error: 'TC_Configure' was not declared in this scope
> 
>    TC_Configure(TC2, 1, TC_CMR_WAVE | TC_CMR_WAVSEL_UP_RC | TC_CMR_TCCLKS_TIMER_CLOCK4); // counter up, /128 divisor
> 
>                                                                                       ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:113:24: error: 'TC_SetRC' was not declared in this scope
> 
>    TC_SetRC(TC2, 1, 2500); // approx 4ms at /128 divisor
> 
>                         ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:114:29: error: 'TC_IER_CPCS' was not declared in this scope
> 
>    TC2->TC_CHANNEL[1].TC_IER=TC_IER_CPCS;
> 
>                              ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:116:32: error: 'NVIC_ClearPendingIRQ' was not declared in this scope
> 
>    NVIC_ClearPendingIRQ(TC7_IRQn);
> 
>                                 ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:117:26: error: 'NVIC_EnableIRQ' was not declared in this scope
> 
>    NVIC_EnableIRQ(TC7_IRQn);
> 
>                           ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:118:18: error: 'TC_Start' was not declared in this scope
> 
>    TC_Start(TC2, 1);
> 
>                   ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp: In member function 'void BaseDMD::end()':
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:123:19: error: 'TC7_IRQn' was not declared in this scope
> 
>    NVIC_DisableIRQ(TC7_IRQn);
> 
>                    ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:123:27: error: 'NVIC_DisableIRQ' was not declared in this scope
> 
>    NVIC_DisableIRQ(TC7_IRQn);
> 
>                            ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:126:28: error: 'NVIC_EnableIRQ' was not declared in this scope
> 
>      NVIC_EnableIRQ(TC7_IRQn); // Still some DMDs running
> 
>                             ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:128:13: error: 'TC2' was not declared in this scope
> 
>      TC_Stop(TC2, 1);
> 
>              ^
> 
> D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2\DMD2_Timer.cpp:128:19: error: 'TC_Stop' was not declared in this scope
> 
>      TC_Stop(TC2, 1);
> 
>                    ^
> 
> Using library DMD2 at version 0.0.3 in folder: D:\Arduino\1- IDE\arduino-1.8.5-windows\arduino-1.8.5\libraries\DMD2 
> 
> exit status 1
> Error compiling for board Generic STM32F103C series.

rtek1000, 2018-05-20:
STM32 Arduino Tutorial - How to use the STM32F103C8T6 board with the Arduino IDE:
https://www.youtube.com/watch?v=K-jYSysmw9w

rtek1000, 2018-05-20:
How to initiallize the two SPI ports of STM32:
http://www.stm32duino.com/viewtopic.php?t=278

h4rm0n1c, 2018-06-21:
I'm sorry, I'm just not very familiar with the STM32, however, like my own addition of esp8266 support was accepted when Angus was the active maintainer, I am in turn willing to review and accept pull requests from those who have expertise on the STM32.

closed 2018-06-21 by h4rm0n1c