freetronics/DMD2 #33: '

arinurcahyo, 2018-06-12:


h4rm0n1c, 2018-06-21:
I am painfully aware of the lack of support for variable width fonts.

I've created a milestone to work towards, shouldn't be too hard, some clever users on the freetronics forums have apparently already figured this out, with their permission I may test and merge their contributions.

arinurcahyo, 2018-06-21:
I found a little solution for my issue using charWidth function.
I had to change this code: 

`if (col == -5) { // -5 used as a buffer when char is hiding after most left column of the display
col = 1; // -5 is used since we use fixed font 5x7
charIndexChanged = true;
charIndex += 1;
}`

into this : 

` if (col == -(thisCharIndexWidth)) {                                                   
      col = 1;                                                         
      charIndexChanged = true;
      charIndex += 1;
      Serial.print("Char Index = "); Serial.println(charIndex);
      thisCharIndexWidth = dmd.charWidth(teks[charIndex]);             
    }`

Now it can scroll any variable width fonts.

However I don't know yet whether the performance is good or not.
At least it fulfill my needs, using 2x2 panel, it can pixel scroll a (very) long text and can scroll 2 text at same time and I didn't see noticeable slowdown.

Look forward to your work.