freetronics/DMD2 #34: Bug with more then 7 Displays in a row

niklasberlin, 2018-09-06:
Hi, I got a Displays which is made out of nine p10 Panels, if i connect my arduino uno and set the SoftDMD to a width of seven panels the first seven displays are working correctly with my test sketch (only a drawLine(0,0,287,0) ). 
The Panels are wired from right to left (viewed from the LED Side) according to the arrows on the Back) Panels are in order: 9, 8, 7, 6, 5, 4, 3, 2, 1 
If i change the width in SoftDMG to 8 only 4 LED's light up, they are located on Panel 1 on the far right in rows 0, 1, 2 and 3 (vertical line, not horizontal line as intended by the drawLine() function)

If i change the width to nine the line is correct on panel 1, 5, 9 and shifted from row 0 to row 1 on panel 2 and 6, on panel 3 and 7 the line is shifted another row to row 2, on panel 4 and 8 the line is shifted to row 3, I hope this helps to find the error. i try to locate it myself but i'm not so experienced with this topic

pererasampath, 2018-12-21:
Hi, I too got the same issue. I am checking DMD2.h file to see whether there are any limitations. but still could not found.

meetpatelvgec, 2019-01-01:
I also got this issue. I had 10 modules in a row and I was using `SPIDMD dmd(10,1);` but it didn't seem to work. As a workaround, I placed the modules in 5 rows and 2 columns and used `SPIDMD dmd(2,5);` which solved my issue.

Note that if you decide to follow this approach, then the digits are located at below (X,Y) cordinates:
1. (0,0)
2. (32,0)
3. (0,16)
4. (32,16)
5. (0,32)
6. (32,32)
7. (0,48)
8. (32,48)
9. (0,64)
10. (32,64)

shr34sachin, 2019-05-25:
I am also having the issue with (8,1) with SPIDMD for esp8266 nodemcu. Upto 7, it works fine. The solution as presented by meetpatelvgec won't work for me since I am using custom font and and that I need to increase the number of columns as (8,1), no other way around.
Did anyone manage to find the bug in the library?
Thank you!

pererasampath, 2019-05-25:
I found a fix for this. The variable that holds the panel dimensions is type byte and when the value exceeds 8 it fails. So I edited DMD2.h or DMD2.cpp ,  do not remember exactly which file and which place and entered the numeric values instead of variable and compiled. it worked. Try out. I'll try to find the exact place and post here.

shr34sachin, 2019-05-27:
Thanks a lot @pererasampath. I will give that a try. In case if you happened to find the exact location within the code, please let me know.

roffy255, 2019-05-28:
i got same issue. After horizontal  series of 7 display, which was working fine, i added one more display in series and updated code SPIDMD dmd(8, 1). It didnt work.  I tried to replace all byte variables to int16_t variables in DMD2.h, DMD2.cpp, DMDFrame.cpp. didnt changed anything.. Still got same issue. @pererasampath, waiting for your post !

shr34sachin, 2019-06-09:
Hi @pererasampath, did you manage to find the places that you have changed? I tried changing datatype of different variables but still couldn't figure that out.
Thank you!

shr34sachin, 2019-06-14:
Finally solved the issue, it was the issue of overflow as has been mentioned by @pererasampath. Need to edit the datatype in the class DMDFrame.
Thanks to @pererasampath for right direction.

ElectronicRescue, 2019-06-15:
@shr34sachin  Are you able to post the code that has to be edited so that this library can be fixed?

shr34sachin, 2019-06-18:
@ElectronicRescue I have posted the code and opened the pull request for the fix. Sorry, I got busy and so can't respond immediately.
Thank you!

pererasampath, 2019-06-18:
Hi, very sorry. I was hospitalized and could not respond timely. Glad @shr34sachin found the and fix the bug. Would like to see the fixed code. Sorry I have no idea of the pull request. Can we see the code?

shr34sachin, 2019-06-21:
@pererasampath Pls check the pull request section, the changes made in two of the files could be found there.

savotech, 2025-06-07:
i have an issue with the display not working with more than 7 panel
i dont know why it cannot be specifically stated here, i worked with @shr34sachin suggestion, it was difficult to find. i got it at last and worked perfectly. this is what i did.

DMDFrame::DMDFrame(uint16_t pixelsWide, uint16_t pixelsHigh)   ----in DMDFrame.cpp (change byte to uint16_t)
DMDFrame(uint16_t pixelsWide, uint16_t pixelsHigh);----- in DMD2.h (change byte to uint16_t)

  const uint16_t width; // in pixels        ----- in DMD2.h (change byte to uint16_t)
  const uint16_t height; // in pixels       ----- in DMD2.h (change byte to uint16_t)

h4rm0n1c, 2026-11-11:
The dimension-overflow fix from #43/#44 is now merged into master through [#72](https://github.com/freetronics/DMD2/pull/72), together with wider stride and bitmap-index arithmetic. Automated framebuffer tests cover the 255/256-pixel boundary, wider frames, copying and scrolling under address/undefined-behavior sanitizers; dimension and pixel checks also passed on ESP32 hardware. The corrected dimension types match the successful fixes reported in this thread. We have not physically tested an eight- or nine-panel chain, and available RAM still limits frame size. Closing the reported overflow bug as fixed.

closed 2026-11-11 by h4rm0n1c