freetronics/DMD2 #49: all led is on and characters is off

MohamadAlsadi, 2019-07-21:
I tried use inverse mode but the charcters not correct

please give me Example INO file with GRAPHICS_INVERSE mode

I am use 1 P10 module

MohamadAlsadi, 2019-07-22:
I try inverse the code in DMD.cpp file 

for(int i = 0; i < rowsize; i++) {
    softSPITransfer(*(rows[3]++), port_r_data, mask_r_data, port_clk, mask_clk);
    softSPITransfer(*(rows[2]++), port_r_data, mask_r_data, port_clk, mask_clk);
    softSPITransfer(*(rows[1]++), port_r_data, mask_r_data, port_clk, mask_clk);
    softSPITransfer(*(rows[0]++), port_r_data, mask_r_data, port_clk, mask_clk);

to :

for(int i = 0; i < rowsize; i++) {
    softSPITransfer(*(rows[3]++), ~port_r_data, mask_r_data, port_clk, mask_clk);
    softSPITransfer(*(rows[2]++), ~port_r_data, mask_r_data, port_clk, mask_clk);
    softSPITransfer(*(rows[1]++), ~port_r_data, mask_r_data, port_clk, mask_clk);
    softSPITransfer(*(rows[0]++), ~port_r_data, mask_r_data, port_clk, mask_clk);

but I get the ERROR :
C:\Users\SADI\Documents\Arduino\libraries\DMD2-master\DMD2.cpp:160:36: error: wrong type argument to bit-complement

LianRave, 2020-01-15:
Could you find a solution for it, I have the same problem , is like the panels that I bought are working with inverse logic ¿?

MohamadAlsadi, 2020-01-15:
Please add me on Whatsapp
+963993597406