freetronics/DMD2 #55: Library is preventing the use of long arrays

mmzeynalli, 2020-05-01:
Recently I do simple project where I send pixel data from RaspberryPi via UART. In this project, I created array of ~1000 length (screens are 3x5, it makes around 960 bytes of data). First I created it dynamically via calloc, and memory allocation failed. Changing array to static did not change anything. Then, I removed array and tried to increase Arduino buffer size instead. However, even though I have enough buffer size, my code even wont print something to the Serial (and I am not talking about RPi - Arduino yet).

P. S. There is the same issue in original DMD library, and I opened issues in both of them

PozitronikTech, 2020-12-10:
I think the wacthdog intervenes because the time during uart sending is too long. For this reason, communication may not be possible. My suggestion, you can try sending piece by piece?

mmzeynalli, 2020-12-12:
The problem is, I can't create large array, before sending/receiving smth. Can creation of 1024 length static array cause wdt reset?

PozitronikTech, 2020-12-18:
I could not say such a thing. You have to try.