freetronics/DMD2 #66: Issue with DMD2 & OTA update

akshaythoke9116, 2022-02-21:


akshaythoke9116, 2022-02-21:
Whenever esp8266 is updating from HTTP-OTA-Server, esp crashes and serial monitor gives the exception.Please support

Exception (0):
epc1=0x40203e90 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0400 end: 3fff0990 offset: 01a0

>>>stack>>>
3fff05a0:  3fffc278 401026c4 3fffc200 00000022  
3fff05b0:  3fff05c0 3ffecd30 3fffc248 4000050c  
3fff05c0:  400043d0 00000030 00000018 ffffffff  
3fff05d0:  400044ab 3fffc718 3fff06c0 08000000  
3fff05e0:  60000200 08000000 00000001 00000000  
3fff05f0:  0000ffff 3fffc6fc 0000005c 00047000  
3fff0600:  00000000 3ffef884 00000800 00000030  
3fff0610:  3fff1654 40104927 3ffed028 00000000  
3fff0620:  4010235d 3ffed028 3ffed788 00000000  
3fff0630:  3fff0ea4 00000000 3fff1654 4022a8a9  
3fff0640:  3ffe9e4c 00000000 00000000 00000030  
3fff0650:  fffffffe ffffdfff 00000010 3fff4984  
3fff0660:  3fff4984 3fff2780 3fff1654 4022ab10  
3fff0670:  3fff1d0c 3fff1d4c 3fff1d8c 3fff1dcc  
3fff0680:  000000c0 0001e509 0000a8c0 4022ba58  
3fff0690:  3fffc200 3ffef748 00000000 40203eb1  
3fff06a0:  3fff2780 3fff277c 00000080 00000006  
3fff06b0:  00000000 00000000 3fffc248 3ffef87c  
3fff06c0:  00000002 4000410f 00000000 4010053d  
3fff06d0:  3fffc718 40004a3c 00000047 00000000  
3fff06e0:  3fffc718 4010560a 00000047 3ffef970  
3fff06f0:  40207d74 00000047 40207d7d 40209314  
3fff0700:  3fff1c94 00000009 3ffef8c8 40208663  
3fff0710:  3ffef8c8 00000000 00000800 40208715  
3fff0720:  3ffeef1c 00000000 3fff0750 4010068c  
3fff0730:  3ffef87c 0000015d 0000015d 3ffeeea4  
3fff0740:  00000000 3ffeee64 3fff2374 402078ef  
3fff0750:  00000000 00000001 3fff232c 40206032  
3fff0760:  00000000 402052f4 3fff232c 402060ad  
3fff0770:  00000000 00000000 00000000 40208c58  
3fff0780:  00000000 3fff232c 3ffeee64 402064bc  
3fff0790:  3fff1c64 0000000f 00000009 40204e9e  
3fff07a0:  40206080 3ffeee88 00000000 40204eda  
3fff07b0:  00000000 00000000 3ffeee88 3fff07d0  
3fff07c0:  0000000d 3ffeee88 3ffeee64 40206be9  
3fff07d0:  00000000 00000000 00000000 000002d1  
3fff07e0:  3ffef87c 000002ff 3fff2e24 0000001f  
3fff07f0:  0000001d 3fff38b4 0000001f 00000018  
3fff0800:  3fff37bc 0000000f 00000000 3fff386c  
3fff0810:  0000003f 00000006 3fff3804 0000005f  
3fff0820:  00000026 3fff0890 3fff0890 3ffe91f4  
3fff0830:  00000000 00000000 3fff0890 00000000  
3fff0840:  3fff2f44 3fff2f48 000000f0 40208af7  
3fff0850:  3ffeeeec 000002ff 3ffeeefc 40206080  
3fff0860:  3ffeeea4 3ffeee88 3fff0890 40208c30  
3fff0870:  00000000 3ffeee88 3ffeee64 3ffe91f4  
3fff0880:  00000001 3ffeee88 3ffeee64 402074c2  
3fff0890:  3fff2ce4 0000002f 00000026 3fff2c4c  
3fff08a0:  0000008f 00000020 3fff2b84 0000001f  
3fff08b0:  0000000f 3fff2c14 0000002f 00000026  
3fff08c0:  3fff288c 0000000f 00000000 3fff2874  
3fff08d0:  0000000f 00000000 3fff285c 0000000f  
3fff08e0:  00000009 3fff29bc 0000000f 00000004  
3fff08f0:  3fff2eac 0000008f 00000000 3ffef970  
3fff0900:  00000001 3ffef734 0004570b 00000000  
3fff0910:  00000000 40205370 0000000f 3ffef970  
3fff0920:  00000000 00000000 3ffeee64 3ffef960  
3fff0930:  000002b8 3ffeee88 3ffeee64 40206323  
3fff0940:  00000031 00000000 000003e8 40202a48  
3fff0950:  00000000 3fff2b5c 00000000 3ffef960  
3fff0960:  3fffdad0 00000000 3ffeee64 40202ff6  
3fff0970:  3fffdad0 00000000 3ffef959 402092e8  
3fff0980:  feefeffe feefeffe 3ffef970 40100718  
<<<stack<<<

ramanc51, 2025-09-29:
> Whenever esp8266 is updating from HTTP-OTA-Server, esp crashes and serial monitor gives the exception.Please support
>
> Exception (0): epc1=0x40203e90 epc2=0x00000000 epc3=0x00000000 [...]

Have you have solution?